google/CFU-Playground #635: Accuracy of perf_get_mcycle() and perf_get_mcycle64() using Arty a7 100T board

bala122, 2022-07-31:
Hi @tcal-x ,
I've tried to profile some cfu-function calls for my accel. design and as well as for a "test-setup". I'm seeing some unusually high cycle times. For ex:- Using the basic template cfu.v file:

```
// Trivial handshaking for a combinational CFU
  assign rsp_valid = cmd_valid;
  assign cmd_ready = rsp_ready;

  //
  // select output -- note that we're not fully decoding the 3 function_id bits
  //
  assign rsp_payload_outputs_0 = cmd_payload_function_id[0] ? 
                                           cmd_payload_inputs_1 :
                                           cmd_payload_inputs_0 ;
```
And using the "test-setup" code in the header file:
```
/*Single cycle cfu test*/
  int start_cfu_comb = perf_get_mcycle();
  int test_3 = cfu_op1(0,1,2);
  int test_4 = cfu_op0(2,1,2);
  int test_5 = cfu_op1(1,3,4);
  int test_6 = cfu_op0(0,4,5);
  int end_cfu_comb = perf_get_mcycle();
  printf("test_3 %d\n", test_3); //Exp: 2
  printf("test_4 %d\n", test_4); //Exp: 1
  printf("test_5 %d\n", test_5); //Exp: 4
  printf("test_6 %d\n", test_6); //Exp: 4
  printf("Time taken (perf ) %d\n", end_cfu_comb- start_cfu_comb);
  //Breaker
  while(1==1);
  printf("Done\n");
```
When running the above and profiling via perf_get_mcycle ( or mcycle 64) on the Arty a7 100T board @ 70 MHz clk rate, I'm getting cycle times around 180 (mcycle64()) and 92(mcycle()) for the above setup. It should ideally be around 4-6 cycles, so is there any issue with perf_get_mcycle() or mcycle64() ?
I'l try using custom counters in the cfu file to check it as well.
Please let me know soon.
Thanks,
Bala.

bala122, 2022-07-31:
An update:
Following is the disassembly with 4 cfu instructions for conventional compute, 1 other cfu instructions (for time profiling). Even with custom counters, Im observing that the time taken  to come from cfu[0,0] instr to cfu[2,0] instr is around  86 cycles which is surprising since I thought the Vexriscv core is pipelined. Does this mean the Vexriscv core ( with default core chosen) is not being properly pipelined when synthesized on the arty a7? Or another possible issue being the RAM not being pipelined properly and hence when fetching instructions from I-MEM it's taking a long time ?
```
40020be8:	00fa858b          	cfu[0,0]  a1, s5, a5
40020bec:	04fa8a8b          	cfu[2,0]  s5, s5, a5
40020bf0:	00400793          	li	  a5,4
40020bf4:	00300913          	li	  s2,3
40020bf8:	02f9090b          	cfu[1,0]  s2, s2, a5
40020bfc:	00500493          	li	  s1,5
40020c00:	0697848b          	cfu[3,0]  s1, a5, s1
40020c04:	00000793          	li	  a5,0
40020c08:	00f7b40b          	cfu[0,3]  s0, a5, a5
```
I got these messages when synthesizing too:
```
INFO: [Synth 8-3971] The signal "VexRiscv/RegFilePlugin_regFile_reg" was recognized as a true dual port RAM template.
RAM Pipeline Warning: Read Address Register Found For RAM tag_mem_reg. We will not be able to pipeline it. This may degrade performance. 
INFO: [Synth 8-5784] Optimized 4 bits of RAM "tag_mem_reg" due to constant propagation. Old ram width 24 bits, new ram width 20 bits.
RAM Pipeline Warning: Read Address Register Found For RAM tag_mem_reg. We will not be able to pipeline it. This may degrade performance. 
RAM Pipeline Warning: Read Address Register Found For RAM mem_1_reg. We will not be able to pipeline it. This may degrade performance. 
RAM Pipeline Warning: Read Address Register Found For RAM mem_1_reg. We will not be able to pipeline it. This may degrade performance. 
RAM Pipeline Warning: Read Address Register Found For RAM data_mem_grain0_reg. We will not be able to pipeline it. This may degrade performance. 
RAM Pipeline Warning: Read Address Register Found For RAM data_mem_grain0_reg. We will not be able to pipeline it. This may degrade performance. 
.
.
.
RAM Pipeline Warning: Read Address Register Found For RAM data_mem_grain12_reg. We will not be able to pipeline it. This may degrade performance. 
RAM Pipeline Warning: Read Address Register Found For RAM data_mem_grain12_reg. We will not be able to pipeline it. This may degrade performance. 
RAM Pipeline Warning: Read Address Register Found For RAM data_mem_grain13_reg. We will not be able to pipeline it. This may degrade performance. 
```
**_If you could suggest some flags or some settings to modify the Vexriscv core and it's RAM to sort of pipeline it a bit better, that would be very helpful_**
In my main basic simple accel design (with one compute unit), I'm experiencing the setup running faster without the basic accelerator than it is with an accel. peripheral surprisingly so, I wanted to ask this. Hopefully, in a large accel design, it will give performance boosts by parallel computes; avoiding memory accesses and also probably mainly reducing the total instructions read _which seems to be the bottleneck here_
Please let me know about this soon.

tcal-x, 2022-08-03:
Hi @bala122 , sorry for the delay; I'll try to take a closer look soon.

tcal-x, 2022-08-03:
FYI @dolu1990

tcal-x, 2022-08-03:
@bala122 , can you provide the following information:
* which CPU variant you're using (or if it's one you created yourself, what is the configuration?   ICache size etc)
* which toolchain (Vivado I assume?) and version you're using (I'm using Vivado 2019.1)

I'm running the snippet you provided to time 4 consecutive CFU instructions, putting it all in a loop, and once the loop is running, it reports 12 cycles elapsed (using $mcycle).

bala122, 2022-08-03:
Hi @tcal-x , 
Im currently using the default soc ( common_soc) without providing flags for modifying any of its parameters.

Yes, Im currently using vivado 2020.2 ( as far as I know its this,  but will confirm later too).
Are you also getting the RAM pipeline warnings that Im getting?

tcal-x, 2022-08-03:
@bala122 , I **DO** have those warnings from Vivado.

The first time the test sequence is executed, 91 cycles elapse (presumably due to I-cache misses).

For each subsequent time, 12 cycles elapse.

Dolu1990, 2022-08-03:
> The first time the test sequence is executed, 91 cycles elapse (presumably due to I-cache misses).

Sound right :)

bala122, 2022-08-03:
Ok, sure @tcal-x  and @Dolu1990 , Thanks for that. Is there any way to change the I-cache parameters- like no. of sets, ways and also if possible its size so that the contribution to I-cache misses to runtime is reduced. Reading instructions from I-Mem or I-cache is a sort-of regular sequential task (ie, fetching instructions one-by-one after another) and I want to mainly focus on DMEM accesses which can be long due to irregular memory accesses that we see usually in conv2d or other functions. So, I want to simplify my problem of profiling the runtime by only dealing with those DMEM accesses first.
Thanks,
Bala.

bala122, 2022-08-08:
Hi @tcal-x  and @Dolu1990 ,
I've tried building a custom cpu variant using sbt and scala with the following makefile command:
```
VexRiscv_CustomNewCfu.v:` $(SRC) Makefile
	sbt compile "runMain vexriscv.GenCoreDefault --dCacheSize 4096 --iCacheSize 16384 --csrPluginConfig linux --cfu true --iBytesPerLine 64 --outputFile VexRiscv_CustomNewCfu"
```
Originally, the wayCount was 1 , since csrPlugins flag was 'all'. This was changed to 'linux' to increase the number of ways in the i-cache , in order to reduce misses. I've even tried to increase the bytes per line to 512(too big actually), but it didn't reduce the runtime at all.

Here's a basic calculation of the codesize I made based on some assumptions:
The part of the code I want to accelerate has around 300 lines in instructions in the disassembly which corresponds to around 300*4 bytes = 1200 bytes of memory in instruction mem. This is around 2.3- 3 cache lines with a 512 byte block size. So, the full code in itself which can fit in within 3 cache lines (using I-cache which has around 4 waycounts ), should infact not give misses after its fetched once- since the full code can fit in the cache.
The surprising part is the profiling results before and after upgrading I-cache
Before upgrade: ~21M cycles
After upgrade: ~21M cycles again.
Please let me know if there could be any other issue causing this- maybe in synthesis/ in generating the i-cache of such size itself.
Thanks,
Bala.

bala122, 2022-08-08:
Please let me know about the above soon.
Thanks,
Bala.

bala122, 2022-08-09:
Hi, @tcal-x and @Dolu1990 ,
I realised my mistake. I think the framework is fine. I had a lot of stores/ loads causing many dcache misses. I had thought they were Icache misses initially.

I'll let you know if I have any further concerns.
Thanks,
Bala.

Dolu1990, 2022-08-10:
Cool :)

tcal-x, 2022-08-10:
Hi @bala122 , thank for posting your findings here, we really appreciate it.

I had a couple of comments in case you suspect I-cache misses again in the future:
* of course, doing some warm-up first to get code into the icache initially is useful, before you start profiling.
* If there is a code address collision forcing icache misses, there are ways to fix certain routines at certain offsets to avoid the collision, using the `linker.ld` script (here's a [commit](https://github.com/google/CFU-Playground/commit/63a91a0173cd4d6a9d8bb5fd91c122113ffa266b) that does some interesting stuff in this area).
* If there are extra block rams on your FPGA, you could make them into an accessible memory region and place the code kernel in that region, so that you are operating directly out of low-latency RAM.   In that case, you'd want to put it in non-cached address space, or maybe just not have an Icache at all.