google/CFU-Playground #652: Colab, conda packing issue with Renode/Verilator: version `GLIBCXX_3.4.29' not found

tcal-x, 2022-08-17:
To reproduce:
* open colab https://colab.research.google.com/drive/187n-kJ8f4TymjmI9QfHA6vmkCxXIuDFw?usp=sharing
* `Runtime` --> `run all`

After a somewhat long compile, we get to this error:
```
Renode, version 1.13.1.2542 (ec3efdf3-202208150124)

(monitor) s @icebreaker.resc ;  uart_connect sysbus.uart
There was an error executing command 'cpu.cfu0 SimulationFilePathLinux @/content/CFU-Playground/proj/proj_template_v/build/renode/libVtop.so'
Error while opening dynamic library: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /content/CFU-Playground/proj/proj_template_v/build/renode/libVtop.so)
Redirecting the input to sysbus.uart, press <ESC> to quit...
>../proj.mk:202: recipe for target 'renode-headless' failed
make: *** [renode-headless] Error 58
```

Any ideas @PiotrZierhoffer ?

tcal-x, 2022-08-17:
@ajelinski , @PiotrZierhoffer , you should be able to access the colab.

ajelinski, 2022-08-19:
Running `make renode` and `make renode-headless` still works in a fresh Docker container with Ubuntu 18.04 so it was quite a riddle what was the root of the problem...

Currently Conda provides a newer `libstdc++` (from GCC v11) than available in Ubuntu 18.04 (from GCC v8) but also forces built ELFs to use Conda's libraries (through `RPATH`) so there shouldn't be such a problem.
Only removing Conda environment or removing Conda's `libstdcxx-ng` package from it should cause it with a typical usage.

However, in Colab VM's environment there's an `LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libtcmalloc.so.4` variable.
It depends on `libstdc++.so` so it has a side effect of forcing host's `libstdc++.so` to be used.

The problem is gone if the variable is unset before running `make renode-headless`:
```
!(source env/conda/bin/activate cfu-common && cd proj/proj_template_v && unset LD_PRELOAD && make TARGET=icebreaker renode-headless)
```
so perhaps it should be a part of the command within `renode` and `renode-headless` targets or it shouldn't be a part of the environment in the first place.

To make `libVtop.so` work with a host's `libstdc++` the environment shouldn't have any newer `libstdcxx-ng` than v8 but it can have an ugly side effect of limiting the version of Verilator and perhaps other tools to some older versions.

PiotrZierhoffer, 2022-08-19:
FYI @proppy - in case you stumble upon similar issues in your notebooks