google/CFU-Playground #655: Non-deterministic synthesis behavior

ShvetankPrakash, 2022-08-23:
We are seeing that the yosys output of logic synthesis is not consistent from run to run. Two designs that are exactly the same can have different logic cell resource consumption from run to run.

tcal-x, 2022-08-23:
It is due to the timestamp built into the LiteX BIOS, which is embedded in the ROM, which does affect Yosys behavior.

To avoid this, use litex option: `--no-ident-version`

The question is, do we want to always use this option in CFU Playground, or let the user add it to `EXTRA_LITEX_ARGS` when they want nondeterministic behavior.

ShvetankPrakash, 2022-08-23:
Maybe we just add this option permanently to dse_framework in cfu playground so that anyone using the framework gets consistent results at least? Otherwise we can leave it to the user to add it in all other cases, perhaps (and we should include some documentation noting this)?

tcal-x, 2022-08-23:
@ShvetankPrakash can you add `--no-ident-version` to your DSE stuff so that you can kick off your runs while we think about whether to make the change globally?

ShvetankPrakash, 2022-08-23:
Sounds good I will add this for now to kick off the runs @tcal-x