google/CFU-Playground #689: Open OCD shows DEPRECATED. Unable to get output.

mamuneeb, 2022-09-22:
Hii @tcal-x 
I was following the setup guide for build.
After running the below command : 
`make prog TARGET=digilent_arty USE_SYMBIFLOW=1 EXTRA_LITEX_ARGS="--variant=a7-100" EXTRA_LITEX_ARGS="--sys-clk-freq 75000000"`
I am getting this : 

make[2]: Entering directory '/home/mamuneebcfu/CFU-Playground/proj/proj_template'
/home/mamuneebcfu/CFU-Playground/scripts/pyrun /home/mamuneebcfu/CFU-Playground/proj/proj_template/cfu_gen.py 
make -C /home/mamuneebcfu/CFU-Playground/soc -f /home/mamuneebcfu/CFU-Playground/soc/common_soc.mk prog
make[3]: Entering directory '/home/mamuneebcfu/CFU-Playground/soc'
Timing check performed only for Vivado.
Loading bitstream onto board
MAKEFLAGS=-j8 /home/mamuneebcfu/CFU-Playground/scripts/pyrun ./common_soc.py --output-dir build/digilent_arty.proj_template --csr-json build/digilent_arty.proj_template/csr.json --cpu-cfu  /home/mamuneebcfu/CFU-Playground/proj/proj_template/cfu.v --uart-baudrate 1843200 --target digilent_arty --sys-clk-freq 75000000 --toolchain symbiflow --no-compile-software --load
make_soc: cpu_variant is full+cfu
Variant "full+cfu" already known.
INFO:S7PLL:Creating S7PLL, speedgrade -1.
INFO:S7PLL:Registering Single Ended ClkIn of 100.00MHz.
INFO:S7PLL:Creating ClkOut0 sys of 75.00MHz (+-10000.00ppm).
INFO:S7PLL:Creating ClkOut1 eth of 25.00MHz (+-10000.00ppm).
INFO:S7PLL:Creating ClkOut2 sys4x of 300.00MHz (+-10000.00ppm).
INFO:S7PLL:Creating ClkOut3 sys4x_dqs of 300.00MHz (+-10000.00ppm).
INFO:S7PLL:Creating ClkOut4 idelay of 200.00MHz (+-10000.00ppm).
INFO:SoC:        __   _ __      _  __  
INFO:SoC:       / /  (_) /____ | |/_/  
INFO:SoC:      / /__/ / __/ -_)>  <    
INFO:SoC:     /____/_/\__/\__/_/|_|  
INFO:SoC:  Build your hardware, easily!
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Creating SoC... (2022-09-22 05:10:57)
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:FPGA device : xc7a35ticsg324-1L.
INFO:SoC:System clock: 75.000MHz.
INFO:SoCBusHandler:Creating Bus Handler...
INFO:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoCBusHandler:Adding reserved Bus Regions...
INFO:SoCBusHandler:Bus Handler created.
INFO:SoCCSRHandler:Creating CSR Handler...
INFO:SoCCSRHandler:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoCCSRHandler:Adding reserved CSRs...
INFO:SoCCSRHandler:CSR Handler created.
INFO:SoCIRQHandler:Creating IRQ Handler...
INFO:SoCIRQHandler:IRQ Handler (up to 32 Locations).
INFO:SoCIRQHandler:Adding reserved IRQs...
INFO:SoCIRQHandler:IRQ Handler created.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Initial SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoC:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoC:IRQ Handler (up to 32 Locations).
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Controller ctrl added.
INFO:SoC:CPU vexriscv added.
INFO:SoC:CPU vexriscv adding IO Region 0 at 0x80000000 (Size: 0x80000000).
INFO:SoCBusHandler:io0 Region added at Origin: 0x80000000, Size: 0x80000000, Mode: RW, Cached: False Linker: False.
INFO:SoC:CPU vexriscv overriding sram mapping from 0x01000000 to 0x10000000.
INFO:SoC:CPU vexriscv setting reset address to 0x00000000.
INFO:SoC:CPU vexriscv adding Bus Master(s).
INFO:SoCBusHandler:cpu_bus0 added as Bus Master.
INFO:SoCBusHandler:cpu_bus1 added as Bus Master.
INFO:SoC:CPU vexriscv adding Interrupt(s).
INFO:SoC:CPU vexriscv adding SoC components.
INFO:SoCBusHandler:rom Region added at Origin: 0x00000000, Size: 0x00020000, Mode: R, Cached: True Linker: False.
INFO:SoCBusHandler:rom added as Bus Slave.
INFO:SoC:RAM rom added Origin: 0x00000000, Size: 0x00020000, Mode: R, Cached: True Linker: False.
INFO:SoCBusHandler:sram Region added at Origin: 0x10000000, Size: 0x00002000, Mode: RW, Cached: True Linker: False.
INFO:SoCBusHandler:sram added as Bus Slave.
INFO:SoC:RAM sram added Origin: 0x10000000, Size: 0x00002000, Mode: RW, Cached: True Linker: False.
INFO:SoCIRQHandler:uart IRQ allocated at Location 0.
INFO:SoCIRQHandler:timer0 IRQ allocated at Location 1.
[... 94 lines cut ...]
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter driver' not 'interface'
DEPRECATED! use 'ftdi vid_pid' not 'ftdi_vid_pid'
DEPRECATED! use 'ftdi channel' not 'ftdi_channel'
DEPRECATED! use 'ftdi layout_init' not 'ftdi_layout_init'
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
DEPRECATED! use 'adapter speed' not 'adapter_khz'

fpga_program
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 25000 kHz
Info : JTAG tap: xc7.tap tap/device found: 0x13631093 (mfg: 0x049 (Xilinx), part: 0x3631, ver: 0x1)
make[3]: Leaving directory '/home/mamuneebcfu/CFU-Playground/soc'
make[2]: Leaving directory '/home/mamuneebcfu/CFU-Playground/proj/proj_template'

According to the setup guide, i must see the chasing LED pattern but its not showing.

I am using Arty A7-100T.

Is there any issues with the clock frequency.

Please help me with this.
Thank you.

tcal-x, 2022-09-22:
Hello @mamuneeb ,

I think the DEPRECATED! message is typical and can be ignored.  Yes, that is unfortunate.   In fact this may be a good reason for us to switch to using https://github.com/trabucayre/openFPGALoader, another open source project.

But, if you don't see the LED pattern, then something is wrong.  I don't have an Arty board handy where I'm at now, so I'll try to reproduce it later.

**OH, I see your problem:**  

> INFO:SoC:FPGA device : **xc7a35ticsg324-1L**.

There can only be one `EXTRA_LITEX_ARGS` definition (only the last one is used).  So you need to replace:
```
EXTRA_LITEX_ARGS="--variant=a7-100" EXTRA_LITEX_ARGS="--sys-clk-freq 75000000"
```
with
```
EXTRA_LITEX_ARGS="--variant=a7-100  --sys-clk-freq 75000000"
```

mamuneeb, 2022-09-22:
Respected Sir,
Thank you for your response.
I will give a try tomorrow and see if this works out.
Also, can i have your mail or some personal point of contact?
I do have plans to work for a long time on Google CFU PLAYGROUND and i may
need your help frequently.
Thanks a lot for your help.

Regards,
M A Muneeb

mamuneeb, 2022-09-23:
The issue is resolved.
Thank you

closed 2022-09-23 by mamuneeb