google/CFU-Playground #692: OSError: Unable to find any of the cross compilation toolchains

mamuneeb, 2022-09-23:
Hii
I completed the Setup Guide Successfully, able to see the Chasing LEDs on FPGA as well as all the Golden Tests passed.
But when i am executing the below command following as given [here](https://cfu-playground.readthedocs.io/en/latest/step-by-step.html#step-2-profile-and-identify-what-to-accelerate) in Step 2,
`make prog EXTRA_LITEX_ARGS="--variant=a7-100  --sys-clk-freq 75000000"`
It shows 
/home/mamuneebcfu/CFU-Playground/scripts/pyrun /home/mamuneebcfu/CFU-Playground/proj/my_first_cfu/cfu_gen.py 
make -C /home/mamuneebcfu/CFU-Playground/soc -f /home/mamuneebcfu/CFU-Playground/soc/common_soc.mk prog
make[1]: Entering directory '/home/mamuneebcfu/CFU-Playground/soc'
Building bitstream for digilent_arty. CFU option: --cpu-cfu /home/mamuneebcfu/CFU-Playground/proj/my_first_cfu/cfu.v
MAKEFLAGS=-j8 /home/mamuneebcfu/CFU-Playground/scripts/pyrun ./common_soc.py --output-dir build/digilent_arty.my_first_cfu --csr-json build/digilent_arty.my_first_cfu/csr.json --cpu-cfu  /home/mamuneebcfu/CFU-Playground/proj/my_first_cfu/cfu.v --uart-baudrate 1843200 --target digilent_arty --variant=a7-100  --sys-clk-freq 75000000 --build
make_soc: cpu_variant is full+cfu
Variant "full+cfu" already known.
INFO:S7PLL:Creating S7PLL, speedgrade -1.
INFO:S7PLL:Registering Single Ended ClkIn of 100.00MHz.
INFO:S7PLL:Creating ClkOut0 sys of 75.00MHz (+-10000.00ppm).
INFO:S7PLL:Creating ClkOut1 eth of 25.00MHz (+-10000.00ppm).
INFO:S7PLL:Creating ClkOut2 sys4x of 300.00MHz (+-10000.00ppm).
INFO:S7PLL:Creating ClkOut3 sys4x_dqs of 300.00MHz (+-10000.00ppm).
INFO:S7PLL:Creating ClkOut4 idelay of 200.00MHz (+-10000.00ppm).
INFO:SoC:        __   _ __      _  __  
INFO:SoC:       / /  (_) /____ | |/_/  
INFO:SoC:      / /__/ / __/ -_)>  <    
INFO:SoC:     /____/_/\__/\__/_/|_|  
INFO:SoC:  Build your hardware, easily!
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Creating SoC... (2022-09-23 04:18:14)
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:FPGA device : xc7a100tcsg324-1.
INFO:SoC:System clock: 75.000MHz.
INFO:SoCBusHandler:Creating Bus Handler...
INFO:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoCBusHandler:Adding reserved Bus Regions...
INFO:SoCBusHandler:Bus Handler created.
INFO:SoCCSRHandler:Creating CSR Handler...
INFO:SoCCSRHandler:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoCCSRHandler:Adding reserved CSRs...
INFO:SoCCSRHandler:CSR Handler created.
INFO:SoCIRQHandler:Creating IRQ Handler...
INFO:SoCIRQHandler:IRQ Handler (up to 32 Locations).
INFO:SoCIRQHandler:Adding reserved IRQs...
INFO:SoCIRQHandler:IRQ Handler created.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Initial SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoC:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoC:IRQ Handler (up to 32 Locations).
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Controller ctrl added.
INFO:SoC:CPU vexriscv added.
INFO:SoC:CPU vexriscv adding IO Region 0 at 0x80000000 (Size: 0x80000000).
INFO:SoCBusHandler:io0 Region added at Origin: 0x80000000, Size: 0x80000000, Mode: RW, Cached: False Linker: False.
INFO:SoC:CPU vexriscv overriding sram mapping from 0x01000000 to 0x10000000.
INFO:SoC:CPU vexriscv setting reset address to 0x00000000.
INFO:SoC:CPU vexriscv adding Bus Master(s).
INFO:SoCBusHandler:cpu_bus0 added as Bus Master.
INFO:SoCBusHandler:cpu_bus1 added as Bus Master.
INFO:SoC:CPU vexriscv adding Interrupt(s).
INFO:SoC:CPU vexriscv adding SoC components.
INFO:SoCBusHandler:rom Region added at Origin: 0x00000000, Size: 0x00020000, Mode: R, Cached: True Linker: False.
INFO:SoCBusHandler:rom added as Bus Slave.
INFO:SoC:RAM rom added Origin: 0x00000000, Size: 0x00020000, Mode: R, Cached: True Linker: False.
INFO:SoCBusHandler:sram Region added at Origin: 0x10000000, Size: 0x00002000, Mode: RW, Cached: True Linker: False.
INFO:SoCBusHandler:sram added as Bus Slave.
INFO:SoC:RAM sram added Origin: 0x10000000, Size: 0x00002000, Mode: RW, Cached: True Linker: False.
INFO:SoCIRQHandler:uart IRQ allocated at Location 0.
INFO:SoCIRQHandler:timer0 IRQ allocated at Location 1.
INFO:SoCBusHandler:main_ram Region added at Origin: 0x40000000, Size: 0x10000000, Mode: RW, Cached: True Linker: False.
INFO:SoCBusHandler:main_ram added as Bus Slave.
INFO:SoC:CSR Bridge csr added.
[... 79 lines cut ...]
    for k, v in export.get_cpu_mak(self.soc.cpu, self.compile_software):
  File "/home/mamuneebcfu/CFU-Playground/third_party/python/litex/litex/soc/integration/export.py", line 92, in get_cpu_mak
    selected_triple = select_triple(triple)
  File "/home/mamuneebcfu/CFU-Playground/third_party/python/litex/litex/soc/integration/export.py", line 90, in select_triple
    raise OSError(msg)
OSError: Unable to find any of the cross compilation toolchains:
- riscv64-pc-linux-musl
- riscv64-unknown-elf
- riscv64-unknown-linux-gnu
- riscv64-elf
- riscv64-linux
- riscv64-linux-gnu
- riscv-sifive-elf
- riscv64-none-elf
- riscv32-unknown-elf
- riscv32-unknown-linux-gnu
- riscv32-elf
- riscv-none-embed
- riscv-none-elf

make[1]: *** [/home/mamuneebcfu/CFU-Playground/soc/common_soc.mk:115: build/digilent_arty.my_first_cfu/gateware/digilent_arty.bit] Error 1
make[1]: Leaving directory '/home/mamuneebcfu/CFU-Playground/soc'
make: *** [../proj.mk:312: prog] Error 2

Please help me with this.

tcal-x, 2022-09-25:
Did you check if the riscv32 tools are in your path?   Type "riscv32" then TAB two or three times to get name completion -- it should either show many options, or nothing.   If nothing, then perhaps you exited the Conda environment accidentally without realizing it.  You need to be in the Conda environment to access these tools -- in this case since you're using Symbiflow, type `make enter-sf` to enter or re-enter the Conda env.

mamuneeb, 2022-09-26:
Hii 
Thanks a lot @tcal-x , i had accidently exited the Conda environment.
The issue is resolved.
Thank you.

closed 2022-09-26 by mamuneeb