google/CFU-Playground #700: Data corruption error (CRC error)

bala122, 2022-10-10:
Hi @tcal-x  and @mithro ,
I'm getting the following error on uploading to the board. This hasn't happened before and I'm not sure of the reason why it has come up now.
```
--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LITEX-TERM] Received firmware download request from the device.
[LITEX-TERM] Uploading /home/shivaubuntu/CFU-playground/ee18b155/CFU_Playground_Gitlab/proj/proj_nn_sparse/build/software.bin to 0x40000000 (1430104 bytes)...
[LITEX-TERM] Upload calibration... (inter-frame: 10.00us, length: 64)
[LITEX-TERM] Upload to device failed due to data corruption (CRC error)

```

I've tried reducing the speed as well as pointed out in some other issue. When I tried inserting --safe, the word "safe" wasn't popping out anywhere and it wasn't working.

`**Please let me know about this as soon as possible.**` I need the board urgently.
Thanks a lot,
Bala.

tcal-x, 2022-10-10:
@bala122 , yes, this is probably related to https://github.com/google/CFU-Playground/issues/657.   What cache sizes are you using?   I think we saw it specifically when there was no dCache.

I'm not sure if I mentioned it explicitly in the other issue, but there is no way to add `--safe` to the `litex_term` line other than cutting and pasting the entire line and adding `--safe` and rerunning.   With `--safe`, the upload speed is MUCH slower.

I have not tried reducing the UART_SPEED, but that may be worth trying.   You would need to set it to the same value (lower than the default 1843200) for both the `bitstream`/`prog` step and the `load` step.

bala122, 2022-10-10:
I'm using around 32B d-cache (16B block), 512B I-cache and 64B L2.
Is the low d-cache a problem?