google/CFU-Playground #722: Synthesis failing out of box

ShvetankPrakash, 2022-10-26:
Steps to reproduce error:
1.Cloned fresh version of Playground repo
2.Ran `scripts/setup`
3.Ran `make enter-sf`
4.Ran `cd proj/proj_template`
5.Ran `make bitstream`
Failed with following error:

```
Executing module `synthesize`:
    [1/3] : Synthesizing sources: ['/home/CFU-Playground/proj/proj_template/cfu.v', '/home/CFU-Playground/third_party/python/pythondata_cpu_vexriscv/pythondata_cpu_vexriscv/verilog/VexRiscv_FullCfu.v', '/home/CFU-Playground/soc/build/digilent_arty.proj_template/gateware/digilent_arty.v']...
[ERROR]: yosys non-zero return code.
stderr:
ERROR: TCL interpreter returned an error: couldn't read file "/home/CFU-Playground/env/symbiflow/xc7/install/share/f4pga/scripts/xc7/synth.tcl": no such file or directory

/home/CFU-Playground/soc/common_soc.mk:114: recipe for target 'build/digilent_arty.proj_template/gateware/digilent_arty.bit' failed
make[3]: *** [build/digilent_arty.proj_template/gateware/digilent_arty.bit] Error 1
make[3]: Leaving directory '/home/CFU-Playground/soc'
../proj.mk:315: recipe for target 'bitstream' failed
make[2]: *** [bitstream] Error 2
make[2]: Leaving directory '/home/CFU-Playground/proj/proj_template'
```
Upstream change that may be causing this?

tcal-x, 2022-10-26:
With the non-SymbiFlow tools, you can just type `make enter`, which will infer the same actions as if you did `make env; make enter`.   However, with SymbiFlow, just typing `make enter-sf` won't get you everything that you need.   You need to explicitly do both `make install-sf; make enter-sf`.

Try that and see if it fixes things.

ShvetankPrakash, 2022-10-27:
Ah yes wasn't aware of this thanks for letting me know! This did indeed work to fix the issue :)

closed 2022-10-27 by ShvetankPrakash