google/CFU-Playground #728: L2 cache parameter description (sub 1KB)

bala122, 2022-10-28:
Hi @tcal-x and @mithro ,
I wanted to know if Litex just removes the L2 cache (ie, 0K) when we use a value of lower than 1024 bytes. Since, Ive noticed that when I put in a very low size like 64B , during liftoff, L2 is shown as 0. Also, in the digilent_arty.py file L2 size is mentioned as some_number*1024 . So, I mainly wanted to know if this is true for all values < 1KB.
_**Please let me know about this soon**_
Thanks,
Bala.

bala122, 2022-10-29:
Hi @tcal-x , any update on this? I need to confirm this soon.
Please let me know whenever is possible.
Thanks,
Bala.

bala122, 2022-10-31:
Also, I've raised an issue in litedram regarding typical port sizes (https://github.com/enjoy-digital/litedram/issues/312).
I also wanted to know what the typical portsizes between L2 and litedram were, since it could potentially dominate when we go to lower L2 sizes.
Thanks,
Bala.

bala122, 2022-11-01:
Hi @tcal-x , any update on this?
Thanks,
Bala.

bala122, 2022-11-04:
Hi @tcal-x , any update on the port sizes. I think I've seen on some config files that it was 32b for arty, however, I just wanted confirmation on this.
Thanks,
Bala.

alanvgreen, 2022-11-04:
Hi @bala122: CFU-Playground doesn't make any specific changes to how LiteX uses LiteDRAM.

I'm not entirely sure what values you'd like to know, but try adding a `breakpoint()` in `third_party/python/litex/litex/soc/integration/soc.py` around line 1584, just before the [`# L2 Cache`](https://github.com/enjoy-digital/litex/blob/879f1b38bca3a209cd73c187749490d1e3a592aa/litex/soc/integration/soc.py#L1587) comment. Like this:

```python
            # L2 Cache
            breakpoint()               # Temporarily add this
            if l2_cache_size != 0:
                # Insert L2 cache inbetween Wishbone bus and LiteDRAM
                l2_cache_size = max(l2_cache_size, int(2*port.data_width/8)) # Use minimal size if lower

```

Then run `make` as usual. From there you can step through the logic to see exactly what LiteX creates and what the size of each thing is.

Here's what I see when I run it (I have not modified any of the parameters, though)

```
$ make enter-sf
make[1]: Entering directory '/home/osboxes/src/CFU-Playground'
source /home/osboxes/src/CFU-Playground/env/conda/bin/activate cfu-symbiflow && bash
$ cd proj/proj_template
$ make TARGET=digilent_arty clean bitstream

   ... Lots of output ...

INFO:SoCBusHandler:main_ram Region added at Origin: 0x40000000, Size: 0x10000000, Mode: RW, Cached: True Linker: False.
INFO:SoCBusHandler:main_ram added as Bus Slave.
> /home/osboxes/src/CFU-Playground/third_party/python/litex/litex/soc/integration/soc.py(1566)add_sdram()
-> if l2_cache_size != 0:
(Pdb) p l2_cache_size
8192
(Pdb)
```

The Python debugger is an extremely powerful tool for exploring the functionality of complex libraries. If you haven't come across it before, there is a [full list of commands on docs.python.org](https://docs.python.org/3/library/pdb.html#debugger-commands) although I mostly tend to use `p` (print), `n` (next) and `c` (continue).

Hope this helps!

bala122, 2022-11-04:
Thanks a lot for that @tcal-x , just a final query. Could you rerun this with 64 and 256 as L2 cache size and let me know what the printed L2 size is ( for the arty A7 100t board)
Thanks,
Bala.

alanvgreen, 2022-11-04:
You'll find it best if you were to run it yourself - that way you'll be changing precisely the parameters you want and getting exactly the information you need.

Is there anything stopping you from trying it?

bala122, 2022-11-04:
Nope, no issues @tcal-x :)
I thought I'd need the board to run it, turns out it compiles without it. Anyway, it seems to be giving 64 and 256 as the L2 sizes itself, since the indiv. port width is 128b.
Thanks a lot for this debugging tool!

tcal-x, 2022-11-04:
Hi @bala122 , it was @alanvgreen helping you there, not me.   Thanks Alan!

I was going to suggest something similar -- that you can add `print()` statements to the LiteX Python code under `third_party/python/` to help figure out what is going on.   The main places where most of the config and assembly code lives is under `litex_boards/litex_boards/{platforms,targets}/digilent_arty.py`, and `litex/litex/soc/`.   

And as Alan mentioned, this is really a question for the LiteX system, not for CFU Playground.   Have you tried asking on the #litex IRC channel?

One more point for understanding the generated system -- you can look at the generated Verilog directly at `$CFU_ROOT/soc/build/<platform.proj>/gateware/<platform>.v`.

tcal-x, 2022-11-05:
@bala122 , maybe you figured it out, but when you specify <1024B L2 cache, the L2 cache IS still there, so it must just be the LiteX Liftoff message that rounds it down to 0kB.

If you want to look at the details, look in the generated verilog and search for `data_mem_grain`... you should see `data_mem_grain0` through `data_mem_grain15`.  It seems there is one for each byte column.

Also look for signals `soc_basesoc_data_port_adr`, and `soc_basesoc_tag_*`.

bala122, 2022-11-05:
Sure, I'll check it out. Thanks @tcal-x , @alanvgreen.