google/CFU-Playground #731: program results are messy when displayed on my terminal

limingxuan-pku, 2022-11-03:
Hello! I follow [setup guide](https://cfu-playground.readthedocs.io/en/latest/setup-guide.html) to build my cfu-playground by using Arty A7 35T  on my VM ubuntu 20.04.
Everything works well except that the information displayed on my terminal is in a mess.
I run the project in `proj/proj_template`, adding `export EXTRA_LITEX_ARGS="--cpu-variant=perf+cfu"` in `Makefile` to turn perf counters on.
Followings are some of my golden test results.
```
 Counter |  Total | Starts | Average |   Raw
---------+--------+--------+---------+--------------
       |     0  |     0  |   n/a   |            0
    1    |      |     0  |   n/a   |            0
    2    |     0  |     0|   n/a   |            0
    3    |     0  |     0  |   n/a   |            0
    4    |     0  |     0  |   n/a   |          0
    5    |     0  |     0  |   n/a   |            0
       |     0  |     0  |   n/a   |            0
    7    |      |     0  |   n/a   |            0
   212M (    212233017 ) ycles total
```
It's obvious that the output missed a lot of things. For example, in `Counter` column, number `0` and `6` were missed. In the last row, `cycles` missed one letter `c`.
I have run several times and the missing parts are different each time, which means it is totally random.
I also try another Arty A7 35T, but got the same result. So, I think it's not due to the board.
How can I address this problem? Thanks in advance!

tcal-x, 2022-11-04:
Hi @limingxuan-pku , yes, this appears to be corruption on the serial link.   I am a bit surprised you're seeing exactly the same thing with a different board.

In any case, the first thing I'd try is reducing `UART_SPEED` from the default 1843200.   Either temporarily modify the value set in `proj/proj.mk`, or specify it on all make commands:
```
make UART_SPEED=115200 prog
make UART_SPEED=115200 load
```

limingxuan-pku, 2022-11-05:
@tcal-x Thanks! I pretty agree with your idea. LGTM. :)

closed 2022-11-05 by limingxuan-pku

tcal-x, 2022-11-05:
@limingxuan-pku please let me know if my suggested workaround fixes the issue.   You can try speeds between 115200 and 1843200 to see where it starts seeing corruption.

limingxuan-pku, 2022-11-05:
> please let me know if my suggested workaround fixes the issue. 

Hi! @tcal-x I found out some interesting results.
When I use VM Ubuntu20.04 in software `Orcale VM VirtualBox`, I try three URAT speeds: 115200, 57600, 38400. It turns out that only when I lower the speed to 38400 can I print the normal text without missing.
However, when it comes to using VM Ubuntu20.04 in software `VMware Workstation Pro 16`, things are different.  Even with default speed 1843200 can the terminal print the correct text.
A little confused result, but that's what I got.

tcal-x, 2022-11-07:
Thanks @limingxuan-pku for following up!    I don't have much experience running CFU-Playground in a VM (other than Linux on ChromeOS).    But correct handling of USB-connected devices is often an issue.