google/CFU-Playground #733: GDB doesn't support CUSTOM0/cfu instruction on disassemble command.

mgaggero, 2022-11-06:
Hi,
    the `disassemble [/r|/m|/s]` command on gdb running against cfu code does not show the CUSTOM0/CFU instruction mnemonic but its machine code representation.

The toolchain is the August 2020 toolchain from freedom-tools.
GDB version is GNU gdb (SiFive GDB 9.1.0-2020.08.2) 9.1.
GCC version is 10.1.0 (SiFive GCC 10.1.0-2020.08.2).

Maybe related to #470.

[image]

tcal-x, 2022-11-07:
@mgaggero yes you are correct on all counts.   We do have a special disassembly script that patches that, but it doesn't help with GDB.    I haven't had time to follow up on #470 and am not sure when I can, but thanks for raising awareness that it would be useful!

tcal-x, 2022-11-09:
@PiotrZierhoffer , @mateusz-holenko , @kgugala -- do you have any experience with this from your other work with RISC-V custom extensions?    See the link in #470.