google/CFU-Playground #736: Can we custom SoC or CPU and how

limingxuan-pku, 2022-11-12:
Though I know we can custom CPU itself and have read the README in `soc/vexriscv`, I have a few questions about it.
First, what does this line: `PREREQUISITE 2: Ensure that you have the VexRiscv source submodule loaded. ` in `soc/vexriscv/README` exactly mean?
Second, I'd like to add a co-processor to CPU through the AXI bus, in order to compare the performance between CFU and the co-processor. Is it possible to do that here?

tcal-x, 2022-11-15:
Hi @limingxuan-pku!

I'm happy to hear that you are interested in customizing the VexRiscv.   The README may be a bit out of date; I will add some updates here.

`PREREQUISITE 2:` just means to load the *git* submodule for VexRiscv (and load it recursively).   The `./scripts/setup` script will do that, or you could manually do it like this:
```
git submodule update --init --recursive third_party/python/pythondata_cpu_vexriscv
```
There are two levels of submodules; CFU Playground loads https://github.com/litex-hub/pythondata-cpu-vexriscv, and that in turn has https://github.com/SpinalHDL/VexRiscv as a submodule.

As loaded in CFU Playground, you will find the VexRiscv source code under `./third_party/python/pythondata_cpu_vexriscv/pythondata_cpu_vexriscv/verilog/ext/VexRiscv/src/main/scala/vexriscv/`.

But, depending on what type of customizations you wish to do, you may be able to use a new mechanism recently added by @ShvetankPrakash.   Let me see if he has examples.

tcal-x, 2022-11-16:
## Customizing the CPU

I'll leave this example of the new functionality.   I just tested this on the Arty A7-35T board.   You can control each parameter independently:
```
make EXTRA_LITEX_ARGS="--cpu-variant=generate+bypass:false+csrPluginConfig:mcycle+dCacheSize:0+hardwareDiv:false+iCacheSize:8192+mulDiv:false+prediction:none+safe:false+singleCycleMulDiv:false+singleCycleShift:false" prog
make EXTRA_LITEX_ARGS="--cpu-variant=generate+bypass:false+csrPluginConfig:mcycle+dCacheSize:0+hardwareDiv:false+iCacheSize:8192+mulDiv:false+prediction:none+safe:false+singleCycleMulDiv:false+singleCycleShift:false" load
```
We don't have a reference manual for the parameterization, but the list of parameters and their default values is found here:  https://github.com/google/CFU-Playground/blob/main/soc/patch_cpu_variant.py#L148-L160

ShvetankPrakash, 2022-11-16:
@limingxuan-pku @tcal-x yes that should work and I will follow-up soon with reference documentation to outline all the exact details and how to do this!

tcal-x, 2022-11-16:
@limingxuan-pku the set of changeable parameters is determined by our top Vex generation script at https://github.com/google/CFU-Playground/blob/main/soc/vexriscv/src/main/scala/vexriscv/GenCoreDefault.scala.  If you are willing to start modifying that script, you are really unlimited in how you can modify the VexRiscv.

tcal-x, 2022-11-16:
## AXI

@limingxuan-pku I haven't really looked into this; but both LiteX and VexRiscv support AXI, so it should be possible to switch to an AXI-based SoC.   I don't think I'll have time to investigate this in the immediate future, howerver.

Possibly you could use a Wishbone-AXI bridge to attach your accelerator to the current Wishbone bus?

limingxuan-pku, 2022-11-17:
>  I did a quick experiment and found it is easy to switch the SoC to using an AXI bus. 

That's really a good news! In a long run, I will try to attach my accelerator to the bus. Right now, I am going to focus on the development of CFU. Thanks for your reply and patience! :)

closed 2022-11-29 by limingxuan-pku