google/CFU-Playground #744: renode error related to conv and software elf

rajsaktish, 2022-11-30:
Hello,

@tcal-x 
@danc86 

I am facing the following error during make renode. 
FYI, riscv32 toolchain has been installed for Ubuntu and added to $PATH. Confirmed this by typing riscv32 (tab, tab, tab)

src/tensorflow/lite/micro/kernels/conv.cc: In function 'TfLiteStatus tflite::{anonymous}::Eval(TfLiteContext*, TfLiteNode*)':
src/tensorflow/lite/micro/kernels/conv.cc:118:34: error: 'ConvPerChannelWithPackedInt4Weights' is not a member of 'tflite::reference_integer_ops'
  118 |           reference_integer_ops::ConvPerChannelWithPackedInt4Weights(
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  CXX conv_common.cc	conv_common.o
  CXX conv_test.cc	conv_test.o
make[3]: *** [Makefile:201: src/tensorflow/lite/micro/kernels/conv.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: Leaving directory '/home/rajsaktish/repos/CFU-Playground/proj/mnv2/build'
make[2]: *** [../proj.mk:233: /home/rajsaktish/repos/CFU-Playground/proj/mnv2/build/software.elf] Error 2
make[2]: Leaving directory '/home/rajsaktish/repos/CFU-Playground/proj/mnv2'

Any help is appreciated.

Thanks
Raj

danc86, 2022-11-30:
I think the `proj/mnv2`  subdirectory is not covered by CI so it's possible that it was broken by a recent update to TfLM.

Try compiling `proj/proj_template` or making your own project based on that one as a template. It should be up to date.

tcal-x, 2022-11-30:
While @danc86 's answer perfectly fits the symptoms that @rajsaktish reported, and in fact we've seen and fixed these ourselves a few times, I have some new data:

* `mnv2_first` is run in CI (by `test-projects.yml`) and is passing
* I just ran the following myself in a new updated sandbox on branch `main`:
```
% cd proj/mnv2_first
% make renode
```
and it worked correctly.

So I think if you update to current latest `main`, then rerun `./scripts/setup`, and then try again, it might work.

Please try and let us know if that fixes it.