google/CFU-Playground #745: accuarcy of function perf_get_mcycle64()

limingxuan-pku, 2022-12-01:
Hello! I have some questions about function perf_get_mcycle64().
There exists some difference in the total cycles if I use perf_counters in contrast to not using them.
The following shows the result without perf_counter when running KWS model.
```
Running kws
.............
"Event","Tag","Ticks"
0,CONV_2D,18426
1,DEPTHWISE_CONV_2D,4340
2,CONV_2D,12430
3,DEPTHWISE_CONV_2D,4290
4,CONV_2D,11224
5,DEPTHWISE_CONV_2D,4094
6,CONV_2D,12518
7,DEPTHWISE_CONV_2D,4397
8,CONV_2D,11369
9,AVERAGE_POOL_2D,90
10,RESHAPE,2
11,FULLY_CONNECTED,18
12,SOFTMAX,12
 Counter |  Total | Starts | Average |     Raw
---------+--------+--------+---------+--------------
    0    |     0  |     0  |   n/a   |            0
    1    |     0  |     0  |   n/a   |            0
    2    |     0  |     0  |   n/a   |            0
    3    |     0  |     0  |   n/a   |            0
    4    |     0  |     0  |   n/a   |            0
    5    |     0  |     0  |   n/a   |            0
    6    |     0  |     0  |   n/a   |            0
    7    |     0  |     0  |   n/a   |            0
    85M (     85231808 )  cycles total
```
However, if I add `perf_enable_counter` and `perf_disable_counter` in `conv.h` to use some perf_counters, the total cycles change.
```
Running kws
.............
"Event","Tag","Ticks"
0,CONV_2D,21499
1,DEPTHWISE_CONV_2D,4239
2,CONV_2D,9217
3,DEPTHWISE_CONV_2D,4193
4,CONV_2D,8028
5,DEPTHWISE_CONV_2D,3992
6,CONV_2D,8771
7,DEPTHWISE_CONV_2D,4709
8,CONV_2D,8350
9,AVERAGE_POOL_2D,90
10,RESHAPE,2
11,FULLY_CONNECTED,18
12,SOFTMAX,14
 Counter |  Total | Starts | Average |     Raw
---------+--------+--------+---------+--------------
    0    |    57M |     5  |    11M  |     57177105
    1    |    33M | 302720  |   108   |     32978671
    2    |     0  |     0  |   n/a   |            0
    3    |     0  |     0  |   n/a   |            0
    4    |     0  |     0  |   n/a   |            0
    5    |     0  |     0  |   n/a   |            0
    6    |     0  |     0  |   n/a   |            0
    7    |     0  |     0  |   n/a   |            0
    75M (     74897487 )  cycles total
```
It's strange that total cycles reduce a lot, I don't think it's due to error because the difference is so huge.
I know the total cycles are counted by using function `perf_get_mcycle64()`, which I think will not be affected by whether to use perf_counters.
Thanks in advance!

alanvgreen, 2022-12-01:
Hi Mingxuan,

In this case it seems that adding performance counters, which should slow
the calculation down, has actually sped it up! This is very interesting.

Since answering your question may require a little investigation, could you
please raise this question as a bug at
https://github.com/google/CFU-Playground/issues/new ?

Please include the following information:

1. Which platform are you working on?
2. What exact command did you run to build the system used to make the
above measurements?
3. Please publish your git repo and send a link to the branch used to run
the above.

Thanks,

Alan

tcal-x, 2022-12-01:
@limingxuan-pku , can I assume that you used CPU variant "perf" or "perf+cfu" in both cases?

We have observed small code changes cause rather large runtime changes.   The seems to be some sensitivity to code placement ---  when adding calls to the perf routines,  they should get inlined, which moves the location of other code, adding or removing L1 I-cache collisions.

You could look at the disassembly in `build/software.elf.dis` to see if there are any large changes in function code placement.

Ideally we'd have counters recording Icache and Dcache misses, but we don't have immediate plans for adding them.

limingxuan-pku, 2022-12-02:
> Please publish your git repo and send a link to the branch used to run the above. 

Hi! @alanvgreen @tcal-x 
Here is my [git repo](https://github.com/limingxuan-pku/my_kws).
I have written detailed README to guide you to recur my issue.
Thanks for your effort!