google/CFU-Playground #752: "make env" is broken

tcal-x, 2022-12-23:
After running `./scripts/setup/`, then `make env`, I see:
```
Preparing transaction: done
Executing transaction: done
/media/tim/GIT/clean/CFU-Playground/env/downloads/Miniconda3-latest-Linux-x86_64.sh: 438: [[: not found
```
That line is
```
    if [[ "${env_name}" == "*" ]]; then
```

I'm seeing this both locally with Ubuntu 20.04 and also in Github Actions (also 20.04?) -- see the failing tests in #751.

tcal-x, 2022-12-23:
FYI @PiotrZierhoffer @proppy

tcal-x, 2022-12-23:
Not sure if this is expected?
```
tim@tim-ubuntu:/media/tim/GIT/clean/CFU-Playground$ ls env/conda/envs/
total 12
4 '*'/  4  ./  4  ../
```

mithro, 2022-12-23:
It looks like the `Miniconda3-latest-Linux-x86_64.sh` installer has *just* started using "bashisms" in their shell script?

The instructions at https://docs.conda.io/projects/conda/en/latest/user-guide/install/linux.html do say `bash Miniconda3-latest-Linux-x86_64.sh`

haaslukas, 2023-01-04:
just started with a fresh ubuntu installation and git cloned the current repo. still getting the error
`/CFU-Playground/env/downloads/Miniconda3-latest-Linux-x86_64.sh: 438: [[: not found`

shouldn't the bash fix be merged already? if not, where is Miniconda3-latest-Linux-x86_64.sh being called in order to add the bash command manually?

proppy, 2023-01-05:
@haaslukas looks like the submodule in https://github.com/SymbiFlow/make-env/tree/59adb0f248cc4a5d764b6b06224bf6adea7fa36e would need to be updated to point to a rev that includes https://github.com/SymbiFlow/make-env/pull/31

as a workaround you should be able to update the submodule manually with:
```
git submodule update --remote
```

PR #754 (2023-01-07): Bump make-env submodule to get bash fix.  Also master-->main.
Fixes #752.
closed 2023-01-07 by tcal-x