google/CFU-Playground #773: `make renode` and `make renode-headless` errors in the *The Step-by-Step Guide to Building an ML Accelerator*

FelipeWcosta, 2023-02-01:
I am trying execute the *The Step-by-Step Guide to Building an ML Accelerator* as follows:
```
make enter
```
```
cp -r proj/proj_template proj/my_first_cfu
```
```
cd proj/my_first_cfu
```
I noticed differences between the `Makefile` file of `proj/example_cfu` and the `Makefile` file of `proj/my_first_cfu` but when I try to run the Renode with `make renode` or `make renode-headless` it's does not working and show me the following error:
* ```
   ERROR: riscv64-unknown-elf-gcc can not compile programs.
  ```

I can run Renode in `proj/example_cfu` without any errors but with above errors when I do it in `proj/my_first_cfu`.

**Observation**: I made the changes between the Makefile files and it still doesn't work.

Thankful for the help!

FelipeWcosta, 2023-02-01:
Recently I try to copy `proj/example cfu` folder to another folder with `cp -r proj/example_cfu proj/test_cfu` and is appearing the same error.

tcal-x, 2023-02-01:
Hi @FelipeWcosta , it seems to be a tool installation problem.   What is your operating system?

From the top directory of CFU-Playground, what do you see under `env/conda/envs/cfu-common/bin`? 

Hmm, do you have a riscv64* toolchain installed somewhere else in your path already?    There may be an issue with that toolchain conflicting with the riscv32 toolchain provided by conda (make env / make enter).

FelipeWcosta, 2023-02-01:
Hi @tcal-x,  thank you for the reply. Yes, I have riscv64 toolchain in my home dictory, following the directory mentioned I can see the `riscv32` toolchain. Can I fix it just with `make env` and `make enter`?

tcal-x, 2023-02-01:
@FelipeWcosta can you try *hiding* your riscv64 toolchain by making the directory unreadable (doing something like `chmod 000 <directory>`)?     Then `make clean` and `make renode` again...

FelipeWcosta, 2023-02-01:
Good news, it's worked. Grateful for your help @tcal-x

closed 2023-02-01 by FelipeWcosta