google/CFU-Playground #781: "make renode" is broken -- due to Verilator 5.006?

tcal-x, 2023-02-23:
After updating renode, and entering the conda env with "make enter", I get an error when attempting "make renode" in any of the project directories:

```
-- Executing Verilator...
Verilator command: "/usr/local/google/home/tcal/google/CFU-Playground/env/conda/envs/cfu-common/bin/verilator_bin --compiler gcc --prefix Vcfu --Mdir /usr/local/google/home/tcal/google/CFU-Playground/proj/proj_template_v/build/renode/CMakeFiles/libVtop.dir/Vcfu.dir --make cmake --cc -y /usr/local/google/home/tcal/google/CFU-Playground/proj/proj_template_v -Wno-WIDTH -Wno-CASEINCOMPLETE -Wno-CASEOVERLAP /usr/local/google/home/tcal/google/CFU-Playground/proj/proj_template_v/cfu.v"
Output:
%Error: Cannot find verilated_std.sv containing built-in std:: definitions:/usr/local/google/home/tcal/google/CFU-Playground/env/conda/envs/cfu-common/share/verilator/include/verilated_std.sv
        ... Looked in:
             /usr/local/google/home/tcal/google/CFU-Playground/proj/proj_template_v//usr/local/google/home/tcal/google/CFU-Playground/env/conda/envs/cfu-common/share/verilator/include/verilated_std.sv
             /usr/local/google/home/tcal/google/CFU-Playground/proj/proj_template_v//usr/local/google/home/tcal/google/CFU-Playground/env/conda/envs/cfu-common/share/verilator/include/verilated_std.sv.v
             /usr/local/google/home/tcal/google/CFU-Playground/proj/proj_template_v//usr/local/google/home/tcal/google/CFU-Playground/env/conda/envs/cfu-common/share/verilator/include/verilated_std.sv.sv
             /usr/local/google/home/tcal/google/CFU-Playground/env/conda/envs/cfu-common/share/verilator/include/verilated_std.sv
             /usr/local/google/home/tcal/google/CFU-Playground/env/conda/envs/cfu-common/share/verilator/include/verilated_std.sv.v
             /usr/local/google/home/tcal/google/CFU-Playground/env/conda/envs/cfu-common/share/verilator/include/verilated_std.sv.sv
             /usr/local/google/home/tcal/google/CFU-Playground/proj/proj_template_v/build/renode/CMakeFiles/libVtop.dir/Vcfu.dir//usr/local/google/home/tcal/google/CFU-Playground/env/conda/envs/cfu-common/share/verilator/include/verilated_std.sv
             /usr/local/google/home/tcal/google/CFU-Playground/proj/proj_template_v/build/renode/CMakeFiles/libVtop.dir/Vcfu.dir//usr/local/google/home/tcal/google/CFU-Playground/env/conda/envs/cfu-common/share/verilator/include/verilated_std.sv.v
             /usr/local/google/home/tcal/google/CFU-Playground/proj/proj_template_v/build/renode/CMakeFiles/libVtop.dir/Vcfu.dir//usr/local/google/home/tcal/google/CFU-Playground/env/conda/envs/cfu-common/share/verilator/include/verilated_std.sv.sv
%Error: Exiting due to 1 error(s)

CMake Error at /usr/local/google/home/tcal/google/CFU-Playground/env/conda/envs/cfu-common/share/verilator/verilator-config.cmake:244 (message):
  Verilator command failed (return code=1)
Call Stack (most recent call first):
  CMakeLists.txt:30 (verilate)

-- Configuring incomplete, errors occurred!
See also "/usr/local/google/home/tcal/google/CFU-Playground/proj/proj_template_v/build/renode/CMakeFiles/CMakeOutput.log".
See also "/usr/local/google/home/tcal/google/CFU-Playground/proj/proj_template_v/build/renode/CMakeFiles/CMakeError.log".
make[1]: *** [../proj.mk:219: renode-scripts] Error 1
```

If I specify an older Verilator package in `conf/environment.yml`, 5.004 or before, it works ok.

These worked:
```
  - litex-hub::verilator=5.004_11_g51de2c919=20221206_125647
  - litex-hub::verilator=5.004_64_gff2f711d0=20230116_180937
```
These didn't work:
```
  - litex-hub::verilator=5.006_99_g45f955f17=20230218_072610
  - litex-hub::verilator=5.006_94_g026bbc306=20230214_175409
  - litex-hub::verilator
```

FYI @PiotrZierhoffer @proppy -- have there been any changes to the conda-eda package for Verilator?

PiotrZierhoffer, 2023-02-24:
Yes, there has been a change to the verilator package, which may or may not be relevant https://github.com/hdl/conda-eda/commit/9dd40b1ddfdc1587a0fb7507a7d2ee9d2811c06c

PiotrZierhoffer, 2023-02-24:
Naive fix attempt: https://github.com/google/CFU-Playground/pull/783

tcal-x, 2023-03-01:
@PiotrZierhoffer , has this been fixed?   I don't see the issue anymore.   I am using the Verilator package `5.006_115_g06661ab67 20230225_164303`.