google/CFU-Playground #799: `make load` on a Kosagi FOMU board

FelipeWcosta, 2023-06-16:
Hello everyone!

I am try to use CFU-Playground to load a minimal `Vexriscv` CPU added of a CFU because the FOMU board does not have resoucers enough to synthesize the full version of `Vexriscv`. Then, first of all I fixed the error that was occur when I was gonna attach the board into my `Ubuntu-22.04` Virtual Machine, for this I just update the firmware with `dfu-util`:
```
 Fomu PVT running DFU Bootloader v1.9.1 -> Fomu PVT running DFU Bootloader v2.0.4
```
Following this [tutorial](https://workshop.fomu.im/en/latest/bootloader.html).

The next step was made the board be available in the USB port, to do that, I just followed this [steps](https://workshop.fomu.im/en/latest/requirements/drivers.html). When I run `dfu-util -l` the log is:
```
dfu-util 0.11-dev

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [1209:5bf0] ver=0101, devnum=3, cfg=1, intf=0, path="1-1", alt=0, name="Fomu PVT running DFU Bootloader v2.0.4", serial="UNKNOWN"
```
It is ok, I can load some hardware descriptions into FOMU board.

Then, I followed the CFU-Playground setup guide running `make env` and `make enter` to set a environment for Lattice FPGA's. I navigate to `proj_template` directory and runned `make TARGET=kosagi_fomu EXTRA_LITE_ARGS="--cpu-variant=minimal+cfu" prog` and the the compilation ends with:
[image]

And I runned `make TARGET=kosagi_fomu EXTRA_LITE_ARGS="--cpu-variant=minimal+cfu" load`, the result was this:
[image]

But, there are no `/dev/ttyUSB0` available, for example, to attach the board:
[image]

And doing `lsusb` I can see the bootloader:
[image]

If someone can help me to fix it, I would appreciate!

FelipeWcosta, 2023-07-11:
The last update, I changed the programmer to use DFU directly on Fomu workflow and worked very well.

tcal-x, 2023-07-11:
Hi @FelipeWcosta , I'm sorry I was not able to help earlier!   Unfortunately your Fomu adventure coincided with 3 weeks of my travel away from keyboard.

I think you discovered the main issue, that the tty appears at /dev/ttyACM0 usually.   Our suggested hack is to specify the TTY on the make line:   `make TARGET=kosagi_fomu  ....  TTY=/dev/ttyACM0  load`.    With this, it should be able to load the bitstream and the software together.   There may be more tips that you find useful at https://github.com/google/CFU-Playground/wiki/Supported-Boards#fomu

We have 'fomu' and 'fomu+cfu' variants that also fit on the Fomu board and may give a bit more performance than the 'minimal' variants.   If you try to use a variant other than 'fomu' or 'minimal', there is some scripting that will automatically substitute a 'fomu' variant for you.   There is a short comment in the LiteX output that scrolls by, so it's easy to miss.

I'm not sure of your current status -- does everything seem to work for you if you use the `TTY=/dev/ttyACM0 ` define?

FelipeWcosta, 2023-07-11:
Hi @tcal-x, I've been working on using CFU-Playground with kosagi fomu board. In a first moment, I find out a problem to attach the board to my VM that was fixed updating the bootloader (in the first comment) and then the `image.bin` was not loaded directly by the fomu workflow script so I did this manually using DFU and some more commands and worked.

Looking at the fomu workflow script I changed to use only the DFU to program the board:
```
prog = dfu.DFUProg(pid='5bf0', vid='1209')
```
Finally, when I run `make TARGET=kosagi_fomu EXTRA_LITE_ARGS="--cpu-variant=minimal+cfu" load TTY=/dev/ttyACM0` it's working fine.

Some considerations, the execution of the person detection model is very slow and considerably increases the number of cycles, I suppose it is because of the minimal version of vexriscv that very possibly has fewer resources.