google/CFU-Playground #805: Renode Error

Halteproblem, 2023-11-01:
When running

scripts/setup
cd proj/example_cfu
make renode

I get following error: 
`
21:34:35.8609 [INFO] Loaded monitor commands from: /home/halteproblem/Dokumente/CFU-Playground/third_party/renode/scripts/monitor.py
21:34:36.4144 [INFO] Including script: /home/halteproblem/Dokumente/CFU-Playground/proj/example_cfu/build/renode/digilent_arty.resc
21:34:36.4279 [INFO] System bus created.
21:34:37.0805 [WARNING] SimulationFilePath not set!
21:34:37.6377 [INFO] digilent_arty: GDB server with all CPUs started on port :10001
21:34:37.7012 [ERROR] cpu.cfu0: Error while opening dynamic library: /home/halteproblem/Dokumente/CFU-Playground/proj/example_cfu/build/renode/libVtop.so: undefined symbol: tlib_try_interrupt_translation_block
`

Halteproblem, 2023-11-02:
If I change the generate_renode_scripts.py to use the antmicro libVtop.so it seems to work.

`    result = """
using sysbus
mach create \"""" + target + """\"
machine LoadPlatformDescription $ORIGIN/""" + target + """.repl
$cfuLinux?=@https://dl.antmicro.com/projects/renode/verilated-ibex--libVtop-s_527064-bf919fbb21bc3242cbc09f05a8a8cac4037daaff
machine StartGdbServer 10001
showAnalyzer sysbus.uart
showAnalyzer sysbus.uart Antmicro.Renode.Analyzers.LoggingUartAnalyzer
"""
    if cfu_lib_filepath:
        result += """
cpu.cfu0 SimulationFilePathLinux $cfuLinux
"""`

tcal-x, 2023-11-03:
@Halteproblem , thanks for the report; I reproduced the issue.    And also replicated the workaround that you provide -- thanks!

The background is that we haven't been able to bump the Renode version for many months due to an unknown issue (https://github.com/google/CFU-Playground/pull/794#issuecomment-1581886463).   The issue here might be that we're mixing the old pinned Renode (which should still work) somehow with more recent Renode (I haven't had a chance to look where that might be happening).

tcal-x, 2023-11-06:
This was caused by a recent change to renode-infrastructure that added a dependence on tlib in `renode_imports.h`:
https://github.com/renode/renode-infrastructure/commit/1858ed0dce8d03e2d1036747b83ab54eff4731bb

`scripts/setup` grabs files from `renode/renode` at a particular commit.  But for files from `renode/renode-infrastructure` it just grabs from current `master`, which is why we were affected and Renode sim broke even though we made no changes in CFU Playground.

I have a PR in progress to pin our reference to `renode/renode-infrastructure` to before the commit that caused the issue.   (Other options that we might consider in the future would be to add a patch for `renode_imports.h`, or, request that upstream they make the 'tlib' dependency conditional using `#ifdef`.)

FYI @PiotrZierhoffer

PiotrZierhoffer, 2023-11-06:
@tcal-x I think pinning is a good way forward. Right now we are in progress of a serious overhaul of the native calls layer, so this might repeat in future

PR #807 (2023-11-13): Pin downloads from renode/renode-infrastructure
Pin downloads from renode/renode-infrastructure
to a particular commit before a change that breaks Renode sim here (issue #805).

This fixes #805.
closed 2023-11-13 by mithro