google/CFU-Playground #808: Symbiflow not working for Xilinx

navan93, 2023-11-23:
Similar to the issue #722 I am seeing that synthesis is failing 

```
Project status:
    [S] bitstream:  bitstream -> /home/navaneeth/Projects/openmpw/CFU-Playground/soc/build/digilent_arty.proj_template/gateware/digilent_arty.bit
    [N] build_dir:  /home/navaneeth/Projects/openmpw/CFU-Playground/soc/build/digilent_arty.proj_template/gateware
    [S] eblif:  synth -> /home/navaneeth/Projects/openmpw/CFU-Playground/soc/build/digilent_arty.proj_template/gateware/digilent_arty.eblif
    [S] fasm:  fasm -> /home/navaneeth/Projects/openmpw/CFU-Playground/soc/build/digilent_arty.proj_template/gateware/digilent_arty.fasm
    [S] fasm_extra:  synth -> /home/navaneeth/Projects/openmpw/CFU-Playground/soc/build/digilent_arty.proj_template/gateware/digilent_arty_fasm_extra.fasm
    [S] io_place:  ioplace -> /home/navaneeth/Projects/openmpw/CFU-Playground/soc/build/digilent_arty.proj_template/gateware/digilent_arty.ioplace
    [S] net:  pack -> /home/navaneeth/Projects/openmpw/CFU-Playground/soc/build/digilent_arty.proj_template/gateware/digilent_arty.net
    [X] pcf:  MISSING
    [S] place:  place -> /home/navaneeth/Projects/openmpw/CFU-Playground/soc/build/digilent_arty.proj_template/gateware/digilent_arty.place
    [S] place_constraints:  place_constraints -> /home/navaneeth/Projects/openmpw/CFU-Playground/soc/build/digilent_arty.proj_template/gateware/digilent_arty.preplace
    [S] route:  route -> /home/navaneeth/Projects/openmpw/CFU-Playground/soc/build/digilent_arty.proj_template/gateware/digilent_arty.route
    [S] sdc:  synth -> /home/navaneeth/Projects/openmpw/CFU-Playground/soc/build/digilent_arty.proj_template/gateware/digilent_arty.sdc
    [N] sources:  ['/home/navaneeth/Projects/openmpw/CFU-Playground/proj/proj_template/cfu.v', '/home/navaneeth/Projects/openmpw/CFU-Playground/third_party/python/pythondata_cpu_vexriscv/pythondata_cpu_vexriscv/verilog/VexRiscv_FullCfu.v', '/home/navaneeth/Projects/openmpw/CFU-Playground/soc/build/digilent_arty.proj_template/gateware/digilent_arty.v']
    [N] xdc:  /home/navaneeth/Projects/openmpw/CFU-Playground/soc/build/digilent_arty.proj_template/gateware/digilent_arty.xdc

Executing module `synthesize`:
    [1/3] : Synthesizing sources: ['/home/navaneeth/Projects/openmpw/CFU-Playground/proj/proj_template/cfu.v', '/home/navaneeth/Projects/openmpw/CFU-Playground/third_party/python/pythondata_cpu_vexriscv/pythondata_cpu_vexriscv/verilog/VexRiscv_FullCfu.v', '/home/navaneeth/Projects/openmpw/CFU-Playground/soc/build/digilent_arty.proj_template/gateware/digilent_arty.v']...
[ERROR]: yosys non-zero return code.
stderr:
ERROR: TCL interpreter returned an error: Yosys command produced an error

make[5]: *** [/home/navaneeth/Projects/openmpw/CFU-Playground/soc/common_soc.mk:115: build/digilent_arty.proj_template/gateware/digilent_arty.bit] Error 1
make[5]: Leaving directory '/home/navaneeth/Projects/openmpw/CFU-Playground/soc'
make[4]: *** [../proj.mk:319: prog] Error 2
make[4]: Leaving directory '/home/navaneeth/Projects/openmpw/CFU-Playground/proj/proj_template'
```
In `digilent_arty_synth.log`

```
-- Running command `tcl /home/navaneeth/Projects/openmpw/CFU-Playground/env/symbiflow/xc7/install/share/f4pga/scripts/xc7/synth.tcl' --
[TCL: yosys -import] Command name collision: found pre-existing command `cd' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `eval' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `exec' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `read' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `trace' -> skip.
ERROR: Can't load module `./xdc': /home/navaneeth/Projects/openmpw/CFU-Playground/env/conda/envs/cfu-symbiflow/bin/../share/yosys/plugins/xdc.so: undefined symbol: _ZN5Yosys7stringfB5cxx11EPKcz
ERROR: TCL interpreter returned an error: Yosys command produced an error
```

Commands I have used are
```
make install-sf
make enter-sf
make prog TARGET=digilent_arty USE_SYMBIFLOW=1
```

tcal-x, 2023-11-29:
Thanks for the report @navan93 !    I haven't been able to reproduce it yet because I hit a different issue trying to do a fresh Symbiflow installation.    

In case it matters, what OS are you using?

navan93, 2023-11-29:
I am using Ubuntu 22.04 desktop.

tcal-x, 2023-12-04:
Hi @navan93 , I have reproduced the error.     That's all I have for now.

tcal-x, 2023-12-12:
This may be related to https://github.com/chipsalliance/yosys-f4pga-plugins/issues/542.   In any case, I find that pinning Yosys to an older version resolves the issue: #809.

tcal-x, 2023-12-15:
I hit the same issue when targeting Nexus FPGAs since that flow also uses the yosys-f4pga-plugins:
```
ERROR: Can't load module `./dsp-ff': /home/tim/CFU-Playground/env/conda/envs/cfu-common/bin/../share/yosys/plugins/dsp-ff.so: undefined symbol: _ZN5Yosys7stringfB5cxx11EPKcz
```
I'll update the PR to pin Yosys in the place used for that flow.

navan93, 2023-12-19:
Thanks @tcal-x for looking into this.

PR #809 (2023-12-20): Pin the Yosys package to a version compatible with the F4PGA plugins
Pin the Yosys package to a version compatible with the F4PGA Yosys plugins.

This fixes #808.

Tested: manually built the bitstream and tested it on an Arty 35T board with the CFU Playground firmware.

When the plugins package updates, we can unpin Yosys.

Also, comment out the download of the xc7a200t database to make the installation of F4PGA faster.

**NOTE**: After you sync to this update, you don't need to reinstall the entire Symbiflow/F4PGA environment.   Just do this (while not *inside* the environment, at the root di [... cut]
closed 2023-12-20 by mithro