google/CFU-Playground #812: warning: memory region `main_ram' not declared

SHIBSIE, 2024-02-28:
Hello, I am new to using CFU-Playground and I was trying to test it running it on one of the boards. I am using the Digilent Basys3 board. I was following the demo Tim Callahan presented in the HPCA 2023, I was able to prog the project onto the board and I could see the blinking sliding pattern; however, when I try to load the project, it presents me with the following error towards the end.

`/home/shibli/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14/bin/../lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld:/home/shibli/CFU-Playground/proj/proj_template_no_tflm/build/ld/linker.ld:20: warning: memory region `main_ram' not declared
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:166: software.elf] Error 1
make[1]: Leaving directory '/home/shibli/CFU-Playground/proj/proj_template_no_tflm/build'
make: *** [../proj.mk:240: /home/shibli/CFU-Playground/proj/proj_template_no_tflm/build/software.bin] Error 2
`
If anyone can help me understand what the issue is I would be very grateful.

tcal-x, 2024-02-29:
NOTICE: DUE TO PERSONNEL CHANGES AT GOOGLE, THIS PROJECT IS NOT CURRENTLY MAINTAINED.

Hi @SHIBSIE , you have done nothing wrong.   It has to do with the fact that the Basys3 board has no extra external memory (in comparison, Digilent's Arty board has a large external memory).  As a consequence, the Litex-boards target file for Basys3 doesn't declare a `"main_ram"` memory region.   This is what causes the error that you see.

Aside: This used to be the case also for the Kosagi Fomu board.   It has a large (128kB) on-FPGA RAM, but it used to be a single `sram` region.   We handled that by modifying the linker script to put all of the sections that typically go into `main_ram` into `sram`.   (When Litex-boards made a change that split the 128kB into 64kB `sram` and 64kB `main_ram`, we had to do another hack to shift all of the memory back into the `sram` region: https://github.com/google/CFU-Playground/blob/main/soc/board_specific_workflows/ice40up5k.py#L74-L77).

Back to the Basys3.   I guess we never attempted putting CFU Playground on it.    The first question is, can we build an on-FPGA RAM large enough for our needs.   For the no-tflm project, I would guess that the answer is yes.   Then, we'd need to make a custom linker.ld script for the basys3.

I think I remember (this was a long time ago when I looked at it) that on Basys3, the FPGA could not access the flash -- the flash could only be used to hold the bitstream to configure the FPGA, but not as extra storage that the FPGA could access.   This is unfortunate since it means we need to put even code and read-only data into our limited on-FPGA RAM.   I think that is why I gave up on trying to use Basys3.

I'll try to find time to dig out my Basys3 board in the next few days, but I cannot promise anything.

SHIBSIE, 2024-03-01:
Hello, Thank you for your quick response. I get what the problem is, I was just using the basys3 board becuase I had it on hand and wanted to test out if everything was working fine, but for my project I will be using abigger board.

Do you have any idea if the ZedBoard would work alright. Thank you for your response and time.

closed 2024-03-05 by SHIBSIE

jahagirdar, 2024-06-21:
I am facing a similar issue with the TUL-Pynq-Z2 board. The board has 512MB of DDR3.
Is there a general guideline on how to create an board specific python file?