google/CFU-Playground #813: /home/shibli/CFU-Playground/soc/bin/litex_term --speed 1843200  --kernel /home/shibli/CFU-Playground/proj/proj_template_v/build/software.bin /dev/ttyUSB0 hangs during load

SHIBSIE, 2024-03-05:
I am working on a Nexys Video board, i've tried multiple boards that kept facing issues, however I got my hands on this board and I am trying to run a simple template on it to check that everything works. Unfortunately, it just hangs and i do not get the menu. If anyone can help me understand what the problem is please.

tcal-x, 2024-03-06:
Hi @SHIBSIE ,
Fortunately I have that board!   I will see if I can reproduce the issue.

tcal-x, 2024-03-06:
This board is different than most in that it seems to have two separate USB / FTDI paths, one for programming and the other for UART.   On boards such as Arty, the same path is used for both.

I'm still working through it myself, and don't yet have the answer for CFU Playground.   But when directly building a LiteX SoC for the board, using just a LiteX installation not CFU Playground, I found that I need to connect the cable to the PROG connector to configure the FPGA, then change the cable to connect to UART to talk to the BIOS running on the FPGA.    It works either by changing a single cable, or by having both connected using a hub (or using multiple USB ports on your computer).   However, when both are connected, you will see both e.g. /dev/ttyUSB0 and /dev/ttyUSB1 --- you might need to try both to find the one that is being used as UART.    

But, assuming you had the cable just connected to PROG, that would explain why you see no response when trying to connect.   

Later I will try to see if I can get CFU Playground to work.

SHIBSIE, 2024-03-06:
Hello @tcal-x  , thank you for your help. I have figured that out during my tests, but still no change. I have tried both ports and tried using a single cable and both, but still ddidnt work. I also checked to see if it is something that has to do with UART speed, but still nothing.

tcal-x, 2024-03-07:
I did get this to also work (with faster UART speed) (and pressing the `CPU_RESET` button):
```
make TARGET=digilent_nexys_video UART_SPEED=921600 TTY=/dev/ttyUSB0 prog load
```
But when I tried with the default speed (even higher), I got garbled text.

SHIBSIE, 2024-03-07:
I have tried this but unfortunately it also did not work. I was able to borrow an Arty A7 board to test with and it worked perfectly, it means the problem was not with my environment as I have suspected. Might be something with the board I don't really know. For now I am going to stick with the Arty.

Thank you for you help, really appreciated.

closed 2024-03-07 by SHIBSIE